OpenCentauri/cc-fw-tools #4: Look into further disabling automatic SW update from ELEGOO

pdscomp, 2025-07-07:
Right now we try to nerf the SW update by adding the host that provides the latest update version to /etc/hosts to redirect to localhost, but there might be some caching of the latest version since I've seen it still offer an update? This may or may not matter (that cache seems to be reset when flashing). Anyway TODO is to verify this isn't a problem or if it is, add more measures to block the auto-update.

hedgehogform, 2025-08-30:
> Right now we try to nerf the SW update by adding the host that provides the latest update version to /etc/hosts to redirect to localhost, but there might be some caching of the latest version since [...]

By changing the /etc/swupdate_public.pem it's literally impossible to update. If they're using the same thing as Anycubic does.

So even if the printer downloads etc it will not install since the checksum will fail if using custom firmware. Just a note.

hedgehogform, 2025-08-30:
Yes I remembered. it doesn't matter if you change the /etc/hosts because the app in Anycubic automatically bypasses it. You will have to replace the string in the `app` binary to stop it.

pdscomp, 2025-08-31:
> Yes I remembered. it doesn't matter if you change the /etc/hosts because the app in Anycubic hardcoded urls to automatically bypasses it. You will have to replace the string in the `app` binary to stop it.

Copy! I figured this might be needed. Mostly just to prevent the annoying popup every time you reboot 😀 Thank you!

suchmememanyskill, 2025-10-17:
Fixed with the new patching system via an iptables block

closed 2025-10-17 by suchmememanyskill